UnexpectedMaker/esp32s3 #7: D5?

gwolber, 2022-12-08:
Hi, I really love the board, congratulations on this great work! 
However, I encountered a problem: After connecting 5V to the USB power pin on the breakout header, my board cannot be powered over the USB port anymore; D5 seems dead for whatever reason. Could you tell me which type of Z-Diode D5 exactly is? 
Thanks, Gerhard

UnexpectedMaker, 2022-12-08:
Oh no! Are you sure you didn't short the pins somehow? You should be able to have the USB connected and the 5V rail connected to external power at the same time.  That's what the back feed diode is there for.
Anyhow, it's a 1A SOD523 Schottky diode, not a Zenner. Any 1A  Schottky will do if you plan on replacing it.

gwolber, 2022-12-09:
Great, thanks!! :)

closed 2022-12-09 by gwolber