UnexpectedMaker/esp32s3 #9: unreliable boot behavior on power up

rottaran, 2023-01-04:
We encountered a timing glitch with one UM FeatherS3 boards. In our hardware, we switch the system on by releasing the Feather's EN pin. Then, quite often the FeatherS3 boots into booloader mode as if the BOOT button was pressed or IO0 pulled to ground. On the oscilloscope it is visible, that the ESP32-S3 chip begins to boot, then activates the internal pullup on IO0. It seems that the voltage is up too late on IO0. 

We solved this by adding a 470k resistor from IO0 to 3.3V. Now booting behaves properly. 
A larger 4.7uF capacitor from RST to ground did not seem to help.

Is this a known issue? It also never appeared when combined with the Feather OLED Display.

UnexpectedMaker, 2023-01-04:
Are you sure it's not IO47 (SPICLK_P) that's triggering the "enter UF2 bootloader mode" ? It's designed to allow for a double reset press to get into UF2 mode. This is part of the UF2 bootloader code.

Otherwise, it could just be a timing thing, and your fix of putting a weak pull-up on IO0 is the correct solution. I leave IO0 floating by design (the recommended implementation by Espressif) and it's a strapping pin and should not be influenced at boot time, but in your case, the pull-up is fine.

A larger cap on RESET wont make any difference, as the RESET pin on the ESP32 is actually an EN pin, so until EN is high, the ESP32 is off, and therefore wont start to influence IO0. So even a 100000uF cap wont help there ;)

rottaran, 2023-01-13:
Thanks for the feedback! In that case, everything works as designed.

closed 2023-01-13 by rottaran

rottaran, 2023-03-03:
Sorry, it's our fault. Due to running out of pins, I attached an 1-wire temperature sensor to the BOOT IO0 pin and forgot the required 4.7k pull-up. Thus, the sensor was delaying the voltage rise during boot and confused the microcontroller.