UnexpectedMaker/esp32s3 #14: FeatherS3 pins for external 32.768kHz crystal

rottaran, 2023-03-03:
While doing experiments with BLE on battery power, we noticed an unexpectedly high current consumption. This makes the battery connector on the board less useful. With dynamic frequency scaling, modem-sleep and CPU pause, we got down to 25mA average. In order to get to the 2 to 5mA light-sleep mode with running BLE connections, it is necessary to connect a 32.768kHz crystal and respective load capacitors to the XTAL_32K pins. 

Sadly, the XTAL_32K seem to be not connected on anything on the UM Feather S3. Maybe it could be a nice feature for future products to at least have unpopulated footprints for the crystal and capacitors on the bottom side of the board.

UnexpectedMaker, 2023-08-03:
Sorry I didn't get around to answering this already :(
IO15/16 that are used for hooking up an external 32K Xtal are already in use on the second STEMMA/QT connector as I2C bus 2, sorry.

rottaran, 2023-08-05:
No problem. I was unaware of the change back then. Seems sensible to separate both I2C busses. Out of curiosity, what speaks against using IO17/IO18 or IO45/IO46 for the STEMMA2 connector?

UnexpectedMaker, 2023-08-06:
Well, IO46 is a strapping IO, so not usable, but ultimately, nothing wrong with other IO for STEMMA2 - but 15/16 is what I chose and it can't be changed now as it's a released design with supporting firmware that initialises those pins as STEMMA/QT #2.

I do custom design works for people needing alterations on my designs - if an external 32k Xtal is something you'd like and it's something you require in quantity, I'm happy to chat about it.

closed 2023-08-09 by UnexpectedMaker